OpenEtherCATsociety/SOES #17: COE problem when SDO is larger than MBSIZE (fragmented MB)

fsabbatini89, 2017-03-01:
Hello Sir,
I found an issue when uploading a OD bigger than the MBSIZE, so fragmented upload has to be done.

Debugging the code i found that after sending the first mailbox response with the "incomplete" bit set, the function ESC_mbxprocess() should return 1 as the ESC_coeprocess() need to keep sending the rest of the OD data.

```
   if (ESC_mbxprocess())
   {
      ESC_coeprocess();
      ESC_foeprocess();
      ESC_xoeprocess();
   }
```

The solution I found was replace the return 0 to the return 1 in the ESC_mbxprocess()

```
   /* outmbx read by master */
   if (ESCvar.mbxoutpost && ESCvar.SM[1].IntR)
   {
      ESC_ackmbxread ();
      /* dispose old backup */
      if (ESCvar.mbxbackup)
      {
         MBXcontrol[ESCvar.mbxbackup].state = MBXstate_idle;
      }
      /* if still to do */
      if (MBXcontrol[ESCvar.mbxoutpost].state == MBXstate_again)
      {
         ESC_writembx (ESCvar.mbxoutpost);
      }
      /* create new backup */
      MBXcontrol[ESCvar.mbxoutpost].state = MBXstate_backup;
      ESCvar.mbxbackup = ESCvar.mbxoutpost;
      ESCvar.mbxoutpost = 0;
      //return 0;
      return 1;   //SOLUTION
   }
```

Regards,
Franco

nakarlsson, 2017-03-01:
Could you please create an pull request?

fsabbatini89, 2017-03-01:
Sure but let me first check if everything works as supposed

nakarlsson, 2017-05-11:
Fixed in 39e2e116d095dbd08cb327d54b0c7bfd721d5a12

closed 2017-05-11 by nakarlsson