OpenEtherCATsociety/SOES #19: Slave Editor generates incompatible code

hsmptg, 2017-03-28:
Hi

Slave Editor generates soes.c code where ESC_write and ESC_read functions have 4 parameters when all the other calls of these functions all over the soes stack have only three parameters!?!

For instance:

soes.c (generated by the "slave editor"): 
void TXPDO_update (void)
{
   ESC_write (SM3_sma, &Rb, TXPDOsize, (void *) &ESCvar.ALevent);
}

esc.h (file from soes stack):
void ESC_write (uint16_t address, void *buf, uint16_t len);

The same happens with ESC_read()!

I am solving this manually editing the soes.c file generated by the "slave editor" but it would be nice to have a compatible code.

Regards,
Helio

nakarlsson, 2017-03-29:
Yes, they are not aligned at the moment, it is on the TODO list.

hsmptg, 2017-11-18:
Dear Andreas Karlsson

I am pleased to see that the SOES code is still being updated (the last commit in Github has only 11 days!) which means that the project is alive and well.
But I notice that the slaveeditor available for download (a crucial tool to generate the application code) is from 2015 and currently is generating inconsistent C code with the last GitHub SOES versions.

As I already pointed out in a previous issue, the ESC_read() and ESC_write() calls in soes.c (generated by slaveeditor) have 4 parameters when those functions in soes/hal/linux-lan9252/esc_hw.c have just 3!
In this case I simply deleted the fourth parameters and solved the problem.

But now with this last version another problem show up: soes.c (generated by slaveeditor) calls sizeRXPDO() that use to be in esc_coe.h/c and in this last version of SOES in Github simply vanished!?!
And other little inconsistencies could be pointed, so my questions are:
- How can I generate code compatible with this last version?
- There is an alternative to the old slaveeditor?
- It is just me who have this problems within the users of SOES?

Regards
Hélio

nakarlsson, 2017-12-14:
New slave editor documentation and instructions can be found here as part of EtherCAT SDK,  https://rt-labs.com/refman/ethercat-sdk/index.php

closed 2017-12-14 by nakarlsson