OpenEtherCATsociety/SOES #20: DC in SOES slave stack

raghurthepro, 2017-03-29:
I already wrote to rt-labs and they said there won't be any support/info on DC in the slave stack. But I thought writing here to a "wider" audience might help. 
I understand that after configuring the DC in the master (PC), it will generate a ARMW signal to the ESC (in my case it's a ET1100 piggyback). This ESC will then generate a SYNC pulse of some length (configured) to my slave board (DSP) through some pin (I am using an interrupt). But what I don't understand is AFTER i get this interrupt, how do I handle the slave stack code? What I think at this moment is to use the interrupt service routine to handle the PDO (input/output data exchange) in the ISR. 
What I want to understand is - Is there something more specific while handling DC in the slave stack? Am I missing some information in my steps? 
Any help with DC in the slave stack will be appreciated. Or atleast a confirmation that my understanding is correct, is also good. 
Kind Regards,
Raghu

raghurthepro, 2017-03-30:
Only now I read the line that says "This organization has no public members" :)

nakarlsson, 2017-03-30:
It is not quite that simple but you are on the right path, you also need state machine validations of the DC settings, and for SPI slaves you need to handle that fact that ongoing SPI transfers might be interrupted by your DC IRQ wanting to read/write processdata.

I'm currently working on a DC sample for XMC4, but that don't need to consider interrupting any ongoing transfers, SPI does due to it address then do consecutive reads/writes.

nakarlsson, 2017-05-29:
SYNC0 sample in-progress to be added

closed 2017-05-29 by nakarlsson