OpenEtherCATsociety/SOES #40: fifo_cnt variable

Sulymarco, 2018-01-29:
Hi everybody,
I don't understand the use of the variable **_fifo_cnt_** in the **_ESC_read_pram_** and **_ESC_write_pram_** functions:
it is loaded, decrement and never used.

According to the LAN9252 datasheet we have to use it to see how many read/write we can perform without needing to check the status again, but this seems not to be the case in the code.

I know that the Microchip LAN9252 SSK code is exactly the same, but this doesn't confort me.

Am I missing something?
Any comment is greatly appreciated.

Marco

Sulymarco, 2018-01-29:
I am referring to the **_esc_hw.c_** file for the LAN9252

Marco

nakarlsson, 2018-01-30:
Seems wierd indeed, have you tried asking Microchip?

Sulymarco, 2018-01-30:
No, not yet.

Sulymarco, 2018-01-31:
I have submitted the issue to the Microchip support.
I will update you about it.

In the meantime any comment is welcome.

Marco

Sulymarco, 2018-02-09:
So far no answer from Microchip support   :-(
@nakarlsson  (or  @everybody else) please can also you submit the question to the Microchip support , just to give them a little bit pressure on the issue?
It would be really interesting to have an answer from them !

Marco

Sulymarco, 2019-03-15:
Finally I received an answer from Microchip (after more than one year)

**_Each FIFO will update at about 800nSec. So, we don’t need to poll the register every time since the PIC is running at a slower speed and polling will reduce the speed even further. If we add code to poll the available count, it will further decrease the throughput. If a high-performance MCU is used it is recommended to use a 1uSec delay or poll the available counter._**

Marco

nakarlsson, 2019-04-26:
Thanks , Ill close this

closed 2019-04-26 by nakarlsson