OpenEtherCATsociety/SOES #53: Large number of PDO outputs

hwstrijker, 2019-04-25:
I'm using a Infineon XMC4300 Relax EtherCAT Kit with SOES. I'm performing different tests using various masters, for one of the test I want to have a large process image. I'm reading a button and have one counter on the slave to update one PDO value, the other PDO's are not used right now.
When I define 696 bits of outputs in the objectfile the master does give the following output:

``
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enxc025e929272d succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 109 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x12 StatusCode=0x001d : Invalid output configuration
Request init state for all slaves
End simple test, close socket
End program
``

The same for the Acontis Ethercat manter:

``
003080 : Master state changed from <UNKNOWN> to <INIT>
003109 : Master state changed from <INIT> to <PREOP>
003117 : ERROR: At least one slave in error status
003117 : Slave error "                    Slave_1001 [0xFFFFFFFF]
                ": - EtherCAT address=1001 - State <PREOP ERROR>(0x12), control status <Invalid Output Configuration>(0x1d)
003117 : Slave init command response error - Slave                     Slave_1001 [0xFFFFFFFF]
                : - EtherCAT address=1001 - Current State change of slave="PREOP to SAFEOP" target state could not be reached
003118 : CEcSlave::ProcessCmdResult() Slave_1001 [0xFFFFFFFF]
                : unprocessed dwInvokeId (0x2)
003119 : ecatSetMasterState() Error 0x98110024 in McSm state 'WAIT_MASTER_SAFEOP' for requested state 'MASTER_SAFEOP_DONE'
003119 : Cannot start set master state to SAFEOP (Result = ERROR: Slave error (0x98110024))
``

however when the output size is 664 bits everything works as expected.
Why do I get this error?
I did some tests using a oscilloscope and it seems the controller does work correctly so somehow the slave stack does have a problem updating this amount of bits.
When using a old version of SOES I don't get the error but the Inputs in the Master do not update however the ESC_read is called by the code. 

With soesv3 form #52 I get the same error code .

Both the object-files are attached, the difference is the PDO length. I did add the slaveinfo as well.  
[objectfile.zip](https://github.com/OpenEtherCATsociety/SOES/files/3116880/objectfile.zip)
[output_slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOES/files/3116861/output_slaveinfo.txt)

nakarlsson, 2019-04-26:
You need to make enough space between SM2 and SM3, it should fit Outputs size *2
Eg. SM2 0x1100 + (PDO size 87 * 3) will get overwritten by Sm3. SOES v3 should generat an ERROR?

hwstrijker, 2019-04-26:
Indeed the SM2 size solved the problem, thanks.

closed 2019-04-26 by nakarlsson