OpenEtherCATsociety/SOES #54: problem with SDO error code

farawayfromhome, 2019-05-28:
Hi.
I want my soes can transfer some errorcode to soem, such as code:0x05030000 ("Toggle bit not changed"). 
And I see code in ecat_slv.c, and I guess I have to set the pointer pre_object_download_hook to a valid pointer. Is this correct? And how can I implement this function?

nakarlsson, 2019-05-29:
Sample of the hook

```
uint32_t my_pre_objecthandler (uint16_t index,
      uint8_t subindex,
      void * data,
      size_t size,
      uint16_t flags)
{
   uint32_t abort = 0;
   if(index == 0x2000)
   {
      int not_ok;
      uint8_t * data_ptr = data;
      not_ok = *data_ptr < 0xFF ? 1 : 0;
      if(not_ok)
         abort = 0x05030000;
   }
   return abort;
}
```

Config hook usage
```
   static esc_cfg_t config =
   {
...
      .pre_object_download_hook = my_pre_objecthandler,
      .post_object_download_hook = NULL,
...
   };
   ecat_slv_init(&config);
```

farawayfromhome, 2019-05-29:
Why use the first byte of Index:2000 to determine if errorcode:0x50300000 is generated?

nakarlsson, 2019-05-29:
Just a dummy example, you wanted to know how you validate SDOs and send an abort.

farawayfromhome, 2019-05-29:
Okay. Thanks a lot.
Is there any possibility that abort: 0x50300000 will happen in COE?

nakarlsson, 2019-05-29:
I persume you've seen it? If you have you should ask the slave provider? Ref the Issue you have onging in SOEM, https://github.com/OpenEtherCATsociety/SOEM/issues/306

nakarlsson, 2019-06-13:
question answered

closed 2019-06-13 by nakarlsson