OpenEtherCATsociety/SOES #55: Problem with SOES on STM32 with LAN9252

imyosha, 2019-05-31:
Hi,
I am trying to run EtherCAT communication between STM32F4 using LAN9252 (via SPI) with the SOES library and computer using examples from the SOEM library. I use the v2.2.0 of SOES library, because this is the latest version compatible with the EtherCAT Slave Editor. Like the hal driver for LAN9252 I use the file **_esc_hw.c_** taken from **_SOES/soes/hal/rt-kernel-lan9252_**. In this file, I replaced only the functions responsible for SPI communication (spi_select, write, read) with the corresponding for STM32 and comment a few lines of SPI initialization (in STM32 this is done differently and elsewhere). I created the project using [user manual from rt-labs page](https://rt-labs.com/refman/ethercat-sdk_refman.pdf), and made minor edits due to the use of STM32. After successfully building the code, I checked the operation of the ESC using EtherCAT Explorer:

- EEPROM write/read works,
- ESC register reading works,
- switching the state machine works (including the OP state),

but

- it is impossible to read the Object Dictionary,
- the Process Image tab is also empty,
- in the console, the message "Lost frame" constantly appears.

Next, I made sure that communication between the MCU and the LAN9252 over SPI works using a debugger. After a successful SPI check, I used the simple_test example from the SOEM library and received the following response (instead of XXX… in program response I had my net card address or something like this):

```
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on \ Device \ NPF_ {XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX} succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments: 1: 0 0 0 0
Request operational state for all slaves
Calculated workcounter 0
Operational state reached for all slaves.
OK: all slaves resumed OPERATIONAL.
OK: all slaves resumed OPERATIONAL.
OK: all slaves resumed OPERATIONAL.
.....
```
and response form slaveinfo example from SOEM:
```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX} succeeded.
Time:  268497.215 MBX slave:1 error:0005 Invalid mailbox header
Time:  268497.216 MBX slave:1 error:0005 Invalid mailbox header
...
lots of the same messages with incrementing time value...
...
Time:  268497.280 MBX slave:1 error:0005 Invalid mailbox header
1 slaves found and configured.
Calculated workcounter 0

Slave:1
 Name:joint
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000977 ID: 00000000 Rev: 00000000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   2 F:00010024 Type:3
 SM3 A:1180 L:   2 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 13 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```
The last message is repeated endlessly in the console, but from the source code "simple_test" it is clear that the program should behave differently. In EtherCAT Slave Editor, I added only one record per input and one record per output (2 bytes each) and in the code I made their rewriting and reading in the "cb_get" and "cb_set" functions.

I suspect that the communication does not work on the Application Layer but slaveinfo test shows that there are problems with mailbox (AL STATUS CODE register has value 0). What could be wrong and what should I try to do to properly configure the ESC?

nakarlsson, 2019-05-31:
Send us the SlaveEditor project and a SOEM slaveinfo Wireshark, support[at]rt-labs.com

The old SlaveEditor was not compatible with LAN9252 due to SPI is configured differently and in conflict with ET1100.

nakarlsson, 2019-06-13:
Fixed , esc_hw driver update

closed 2019-06-13 by nakarlsson