OpenEtherCATsociety/SOES #71: DAve + EtherCAT Sdk XMC4800 code generation failed

aeroramesh, 2019-10-31:
Hi 

Am just following the instruction as in https://rt-labs.se/content/uploads/EtherCAT_SDK_DAVE.pdf. 
But when generating xml or code generation of Slave from EthercatSDK .. Its doesnt produce _Slave.c and _slave.h 

and also utype.h is almost empty compared to the one described in 
http://download.rt-labs.com/ethercat/sdk/Infineon/dave_xmc48_soes_sample.zip

Build of entrire project fails,
even tried to open the example as it is .. still same issue 

Thanks in advance for your help 

/Rameshkumar 
Got-S

nakarlsson, 2019-11-04:
Hi,

The example and tutorial are not aligned with current SOESv3. We'll have to update the links.
Sorry for that and thanks for reporting the issue. Send a mail to support[a]rt-labs.com and we send give you updated tutorial and samples.

regards
Andreas

nakarlsson, 2019-11-17:
https://rt-labs.com/infineon/ have been updated with new tutorial and samples. Please close this Issue if the problems have been fixed.

closed 2019-11-18 by aeroramesh