OpenEtherCATsociety/SOES #72: slave stack code

KhloodAmin, 2020-03-01:
I am currently trying to implement a code to allow ethercat communication between a SOEM running on linux and XMC4800 , i don't have the slave stack code tool , Is SOES do the same functionality or i must have the tool to generate the needed files ? as i understand there are 3 layers of stack (Hal / PDI - Generic EtherCAT stack and the application ) , which pare of them is the SOES ? 
Thanks

AAGandomi, 2020-03-01:
Hi, I'm not a project's contributor but I have previously worked with SOES and I believe I might help to answer your questions.

1. "Is SOES do the same functionality or i must have the tool to generate the needed files ?"

For the sake of clarity: the tool you mentioned is a tool given free of charge to ETG members. This product allows to customize the ET9300 code and generate the scaffolding for some of the most common slave applications.
As far as I know, SOES does have a similar tool in the form of an Eclipse plugin/feature, developed and licensed by rt-labs. There might be other alternatives out there, of which I'm not aware.

2.  "as i understand there are 3 layers of stack (Hal / PDI - Generic EtherCAT stack and the application ) , which pare of them is the SOES ?"

EtherCAT slave application are typically structured into 3 layers and the code in this repository covers all of them to some extent.
The project offers a few HAL implementations, the basic features of a "Generic EtherCAT stack" and some slave application's examples to use as a reference.
The codebase is quite simple and I warmly suggest you to give a good look at the code/docs, especially if you want to get a better understanding of its capabilities and ensure that it meets your requirements/expectations.

KhloodAmin, 2020-03-01:
Hi , 
thank you for your help , i couldn't get the ETG membership as they don't offer it for individuals so i have to find another way to write slave stack .
So if i have SOES i need to write the other 2 layers , hal and application  ?

AAGandomi, 2020-03-01:
It really depends on what are your system requirements. I think the XMC4 HAL in the project should suffice for a basic slave application (XMC4800 should be part of the same family). The application layer has to be implemented regardless of the stack: that's where you are supposed to define your slave's control functions/tasks.
Best of luck.

nakarlsson, 2020-03-02:
What platform are you using?

KhloodAmin, 2020-03-03:
> It really depends on what are your system requirements. I think the XMC4 HAL in the project should suffice for a basic slave application (I'm quite sure XMC4800 is part of this family). The application layer has to be implemented regardless of the stack: that's where you are supposed to define your slave's control functions/tasks.
> Best of luck.

Thank you

KhloodAmin, 2020-03-05:
I use Dave and i have XMC4800 , the thing i want to get the ADC values not the LED/Button , as i understand i just have to do the mapping to the ADC register values , any recommendations where i can start ?

nakarlsson, 2020-03-05:
We have a tool you can evaluate for 30days; it can help you develop a ADC based slave instead of the sample LED ones.

Check this tutorial: https://rt-labs.se/content/uploads/EtherCAT_SDK_DAVE.pdf

nakarlsson, 2020-03-12:
@KhloodAmin , Can I close this issue?

closed 2020-03-26 by nakarlsson

Do0Nnie, 2020-07-03:
> We have a tool you can evaluate for 30days; it can help you develop a ADC based slave instead of the sample LED ones.
> 
> Check this tutorial: https://rt-labs.se/content/uploads/EtherCAT_SDK_DAVE.pdf

Link is dead, any update is appreciated

hefloryd, 2020-07-07:
> > We have a tool you can evaluate for 30days; it can help you develop a ADC based slave instead of the sample LED ones.
> > Check this tutorial: https://rt-labs.se/content/uploads/EtherCAT_SDK_DAVE.pdf
> 
> Link is dead, any update is appreciated

The link should now work again.