OpenEtherCATsociety/SOES #73: Setting Serial Number

vostrenkov, 2020-04-24:
Is there any way to set Serial Number in SDO structure by slave itself during initialization?
I have several devices with same configuration and i have to manually edit objectlist file for each device to set such parameters like serial number and station alias.
It would be nice to get them calculated with reference to MCU unique ID, but i can't get if I can rewrite those parts of SDO manually before initialization.

nakarlsson, 2020-04-30:
How do you handle the EEPROM versions of that data?

I guess, the SDO could be populated by reading the EEPROM value and set the SDO, given it is not const. Otherwise you'll have a different values in OD and EEPROM?

vostrenkov, 2020-04-30:
For now i generate EEPROM data and sources using rt-labs EtherCAT SDK. I can not set serial number from the UI, this field is grayed out. So i have only one version of EEPROM .bin file per device type.
 
I guess i can write EEPROM from my controller (stm32 + lan9252) to set same values in EEPROM and OD. But how can I write serial number value in SDO? Do i have to change sources to make Objects structure non const?

nakarlsson, 2020-05-04:
We'll improve the serial number handling in our EtherCAT SDK, but it doesn't make sense to set the serialnumber from the UI. It is typically set during production and the EEPROM value should be the same as the OD 0x1018:04, hence it would be best to set the OD value based on the EEPROM value.

yes, either the structure would need to be non const, or maybe the prefered way, add a pointer to a uint32_t value to be populated with the EEPROM value.

ETG 1000.6 , SII/EEPROM word address 0xE should be the same as OD 0x1018:04

vostrenkov, 2020-05-04:
Got it, thank you!

nakarlsson, 2020-06-04:
In a coming EttehrCAT Slave Editor release will add a serial variable for this puspose. The application will need to set it still, otherwise it'll be 0x0.

closed 2020-06-04 by nakarlsson