OpenEtherCATsociety/SOES #74: Not passing from ESC_checkSM23

viniciusmiguel, 2020-04-30:
Hello.
I'm implementing the lib on a STM32F4 using the LAN9252.
I generated the EEPROM and source files using your plugin for eclipse.

I cannot change the state from PRE OP. 
Debugging the application i was able to identify the error is caused in the ESC_checkSM23 function.
where this validation
 !(SM->ActESC & ESC_SM2_act)`
 is causing the problem. the remaining validations:
`(etohs (SM->PSA) != ESC_SM2_sma) || (etohs (SM->Length) != ESCvar.ESC_SM2_sml)
       || (SM->Command != ESC_SM2_smc)` 
are correct.

I'm using TwinCAT to test my slave.
Could you give a hint what could be the cause for  ActESC to fail.

nakarlsson, 2020-05-04:
What values do you have in SM->ActESC and ESC_SM2_ac?

nakarlsson, 2020-06-04:
@viniciusmiguel: Can we close this?

nakarlsson, 2020-07-01:
inactive

closed 2020-07-01 by nakarlsson