OpenEtherCATsociety/SOES #78: ZYNQ7000: FoE: missing last 3 byte in each packet

shinnoyishi, 2020-07-15:
Hello,

I am running SOES on ZYNQ7000 board, 
It can enter operation mode and everything looks fine.
But when I try to test FoE function, I found that the last 3 byte in each packet is always missing. The rest are correct.
Any idea?
Thank you in advance.

nakarlsson, 2020-07-16:
Please post a wireshark and perhaps a sample file that fails?

shinnoyishi, 2020-07-23:
Hi nakarlsson,

Attached is the result of Wireshark.
I compared it also with Beckhoff SSC stack.
FoE_SSC is the one tested with SSC stack.
Check packet 307 the data is correct.
Read data from IP core is also correct.

FoE_SOES is the one tested with SOES stack.
Check packet 323 the data is also correct.

But If I try to read the data from IP core, it seems not correct from SOES stack

The last 3 byte is 0 0 0 while it should be ff ff ff.

Thank you!

And as a background:

We are using EtherCAT IP core from Beckhoff in FPGA side and stack on ARM side on ZYNQ 7020.
The IP core is verified with SSC stack.

nakarlsson, 2020-07-23:
I think you need to attach the files at the Github Issues, not via mail, I can't see them anyway. Can you share your sample application and configuration in some way? It's possible to send to support[at]rt-labs.com

Please clarify where this missing 3 bytes are detected, function and line in SOES. Just reading from IP is too vague. Does that mean in ESC_read/write or in Mailbox read/write or in FOE write or FOE read function.
"
But If I try to read the data from IP core, it seems not correct from SOES stack
The last 3 byte is 0 0 0 while it should be ff ff ff.
"

nakarlsson, 2021-01-05:
can we close this? we've other customers using Zynq that don't experience this problem using FoE.

shinnoyishi, 2021-01-09:
Hi,

Yes, please close it.
Thank you.

closed 2021-01-09 by nakarlsson