OpenEtherCATsociety/SOES #84: Using SOES or modifying the SOEM

stevedanomodolor, 2020-08-28:
I am currently trying to use an old ethercat driver that initially was working with the old **ros_ethercat_eml** master implementation. I want to change and use soem. When using the slaveinfo to read the drive info, I get the following: 
The extra information I printed was to see at what stage the values were configured. 

Slave 1: old EtherCat slave
slave 2: new EtherCat slave
slave 3: new EtherCat slave

```

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
Number of slaves: 3
**********Debugging information(Before mapping configuration)**********

***********************************Slave 1 info***********************************
slave 1 ->Driver state 17
slave 1 ->Configured address: 1001
slave 1 ->output bit: 0
slave 1 ->output bytes:0
slave 1 ->startbit in first output byte 0
slave 1 -> input bit:0
slave 1 -> input bytes:0
slave 1 ->SM: 0: Start Address: 0x0
slave 1 ->SM: 0: Length: 0x0
slave 1 ->SM: 0: Flags: 0x0

slave 1 ->SM: 1: Start Address: 0x0
slave 1 ->SM: 1: Length: 0x0
slave 1 ->SM: 1: Flags: 0x0

slave 1 ->SM: 2: Start Address: 0x0
slave 1 ->SM: 2: Length: 0x0
slave 1 ->SM: 2: Flags: 0x0

slave 1 ->SM: 3: Start Address: 0x0
slave 1 ->SM: 3: Length: 0x0
slave 1 ->SM: 3: Flags: 0x0

slave 1 ->FMMU: 0:  Logical start address: 0x0
slave 1 ->FMMU: 0:  Logical length: 0x0
slave 1 ->FMMU: 0:  Logical StartBit: 0x0
slave 1 ->FMMU: 0:  Logical EndBit: 0x0
slave 1 ->FMMU: 0:  Physical start address: 0x0
slave 1 ->FMMU: 0:   Physical StartBit: 0x0
slave 1 ->FMMU: 0:  FMMUtype: 0x0
slave 1 ->FMMU: 0:  FMMUactive: 0x0
slave 1 ->FMMU: 0:  unused1: 0x0
slave 1 ->FMMU: 0:  unused1: 0x0

Slave 1: old EtherCat slave
slave 2: new EtherCat slave
slave 3: new EtherCat slave
**********Debugging information(Before mapping configuration)**********

***********************************Slave 1 info***********************************
slave 1 ->Driver state 17
slave 1 ->
slave 1 ->FMMU: 1:  Logical start address: 0x0
slave 1 ->FMMU: 1:  Logical length: 0x0
slave 1 ->FMMU: 1:  Logical StartBit: 0x0
slave 1 ->FMMU: 1:  Logical EndBit: 0x0
slave 1 ->FMMU: 1:  Physical start address: 0x0
slave 1 ->FMMU: 1:   Physical StartBit: 0x0
slave 1 ->FMMU: 1:  FMMUtype: 0x0
slave 1 ->FMMU: 1:  FMMUactive: 0x0
slave 1 ->FMMU: 1:  unused1: 0x0
slave 1 ->FMMU: 1:  unused1: 0x0

[... 404 lines cut ...]
                       true); // Enable

  start_address += base_status;

  sh->set_fmmu_config(fmmu);

  EtherCAT_PD_Config *pd = new EtherCAT_PD_Config(4);

  // Sync managers
  (*pd)[0] = EC_SyncMan(COMMAND_PHY_ADDR, command_size_, EC_BUFFERED, EC_WRITTEN_FROM_MASTER);
  (*pd)[0].ChannelEnable = true;
  (*pd)[0].ALEventEnable = true;

  (*pd)[1] = EC_SyncMan(STATUS_PHY_ADDR, base_status);
  (*pd)[1].ChannelEnable = true;

  (*pd)[2] = EC_SyncMan(WGMailbox::MBX_COMMAND_PHY_ADDR, WGMailbox::MBX_COMMAND_SIZE, EC_QUEUED, EC_WRITTEN_FROM_MASTER);
  (*pd)[2].ChannelEnable = true;
  (*pd)[2].ALEventEnable = true;

  (*pd)[3] = EC_SyncMan(WGMailbox::MBX_STATUS_PHY_ADDR, WGMailbox::MBX_STATUS_SIZE, EC_QUEUED);
  (*pd)[3].ChannelEnable = true;

  sh->set_pd_config(pd);
```

nakarlsson, 2020-09-01:
Is "Slave 1: old EtherCat slave" a SOES slave?  Anyway, it seems it is not implemented to be conformant, seems to be missing the EEPROM data hence the need to manually configure it from the eml master. SOEM rely on having the EEPROM data.

So you have 2 options.

Fix the SOES(?) slave
Or
Port the manual configuration steps from the eml master to SOEM, the closest available in SOEM is using the configlist that handle missing/erroneous data.

stevedanomodolor, 2020-09-01:
It is one of the drivers from the old pr2 robot. I think it is not a soes
slave and it was configured differently.
Is it still possible to use soes to configure the eeprom data and fix the
slave?

When you mean configlist, you are referring to this right?
https://github.com/OpenEtherCATsociety/SOEM/blob/master/soem/ethercatconfiglist.h,
we have to input the configuration data here right?

nakarlsson, 2020-09-01:
Yes, that configlist.

stevedanomodolor, 2020-09-01:
Thank you for your response,

nakarlsson, 2020-09-23:
@stevedanomodolor , can we close this?

stevedanomodolor, 2020-09-23:
Yes, THANK YOU

closed 2020-09-23 by nakarlsson