OpenEtherCATsociety/SOES #88: Is there a port for 16 bit micros

lowkin18, 2020-09-23:
Just wondering if their is a 16 bit minimum data size version of SOES or if there is any plans to make one in the future?

nakarlsson, 2020-09-30:
Currently there are no plans to support '16 bit minimum size'. At rt-labs we've not had any projects requiring that so far. If you're interested in contributing please feel free. rt-labs can support adding the 16bit min size support as a professional service project.

To answer your questions, yes SDOs are written/read to/from the mailboxes. 
And I think you can have multiple issues causing a faulty behavior due to what this topic is about, there are several "HW" structs that map 1:1 to Byte size fields, _MBX; being one.

lowkin18, 2020-10-04:
Does a mailbox length of 10 seem correct when running slaveinfo from master, seems odd that all the other information seems correct but I am below the COE_MININUM_LENGTH
I am trying to find the problem, and I am finding the documentation on ethercats mailbox protocol for COE specifically hard to track down. 

In terms of contribution, I don't really know what that entails, would I just make a pull request when I get it working with 16 bit alignment?

The project I am currently working on is just a hobby project that I am continuing from my Electrical Engineering capstone, So I don't have the budget to pay for professional support or consulting.

nakarlsson, 2020-10-05:
Take a Wireshark and you'll be able to match/map the protocol(s) data for the different parts of the Frame/Mbx header/CoE header.  Btw, does etohs evaluate to something else than for a little endian target?

```
#define htoes(x) (x)
#define htoel(x) (x)

#define etohs(x) htoes (x)
#define etohl(x) htoel (x)
``` 

Contribution: 
Yes, you'd need to sign a CLA "https://openethercatsociety.github.io/cla/cla_soem_soes.pdf" and provide a pull request,