OpenEtherCATsociety/SOES #93: DC Clock implementation on STM32F4 and LAN9252

AlexCorrado, 2020-11-13:
Goodmorning to all,
I'm using a Pc with Ubuntu as Ethercat master and an ethercat slave which uses a LAN9252.
The slave is connected to a pc with window which is used to load the SOES program and once the software has been loaded, it remains connected in debug mode for checking the variables. I'm using SOEM in Ubuntu 16.04 ROS kinetic and I'm working with the SOEM package developed here. The SOES run on a STM32 F4 and we have done a personalized driver for the LAN9252. I need to implement the DC clock in the slave in order to synchronize the communication between master and multiple slaves. Someone can help me?

Thanks.

mictas, 2020-12-29:
Hello,
I've a similar problem, but I haven't solved it yet.
Could someone help us?
Thank you

nakarlsson, 2021-01-05:
A first step is to enable the SYNC0 interrupt in the LAN9252 and make sure you receive it in in the STM32 F4, You can use SOEM to activate/enable the SYNC.

Check the LAN9252 datasheet for details on the SYNC0 interrupt pin, check the STM32 how to connect an external IO interrupt?

Use SOEM ethercatdc module to setup and activate DC in the slave.

nakarlsson, 2021-05-24:
No activity

closed 2021-05-24 by nakarlsson