OpenEtherCATsociety/SOES #96: Writing SDO from slave

Gregwar, 2021-02-09:
Hello,

Is there a process to publish a value to a read-only SDO variables from the slave using SOES?
If I understand correctly, this value will only be read by the master when starting right?
Is it possible to publish values with lower frequency instead of including it in all PDO transaction?

nakarlsson, 2021-02-10:
Hi,

There are no API/function to post/publish SDOs. The SDO traffic is initiated and run from the master. Other mailbox protocols post/publish data but it is still up to the master if/when it will read the data, EoE for example would post an ARP frame on NIC/IF link up as a Mailbox transfer Slave->Master. When that happens, the slave write data to the Slave->Master mailbox which will signal via ESC SyncManager registers that there is something to read for the Master. It is up to the Master how it polls for that information.

The Master can read SDOs at any given time, as long as the Mailbox communication is up and running and that the SDO access rights allow it. You can publish your data to the SDO in the pre/post upload SDO hook to have the latest SDO value read by the Master.

The Master can poll SDOs at a lower rate than the PDO cycle. You can easily try this out using SOEM.

Gregwar, 2021-02-10:
Hello,
Thanks for your reply.

closed 2021-02-10 by Gregwar