OpenEtherCATsociety/SOES #101: [question] Multiple ECAT Slaves PDO assignments using the SSC Tool

leadermik, 2021-08-04:
Hello again,

 I am trying to create a multi slave system and I am in the process of designing the PDO for each node (inverter system). The SSC Tool in combination with an excel Sheet are being used. 
 
 Currently the process is very long and error prune. A seperate excel Sheet must be created for every slave => parsed => make some minor code updates => flash the target.
 
 As I am new to this 😞, I presume that there must be an easier way to make this happen. For example write the excel is such way that the PDO are getting automatically incremented or ...? 
 
 The setup uses your own SOEM implementation on a [Teensy 4.1](https://www.pjrc.com/store/teensy41.html) board an a ECAT Master and the [F2838](https://www.ti.com/tool/TMDSCNCD28388D) TI Board as an ESC. 
 
 I would be very glad if you cound point me to a some documentation or make a suggestion on how to make this possible 🥇 
 
 Many thanks 😃

nakarlsson, 2021-08-04:
I guess can get insperation from either 
CiA402 and how it support different drives profiles 
Or
The 50x modulare device profiles, the different PDOs can be different moduler?

leadermik, 2021-08-07:
Thank you for your answer. I have been searching for material / examples, but with little luck.

I am not 100% whether I am understanding it correctly. I have found an sample excel sheet where the module number nn can be set. Yet I do not know where it must be set. Moreover when it is set the SSC Tool will generate the rest of the messages? 

Considering the CiA402. Could you provide some examples? Maybe some links for the CiA402 or the 50x Modular Decive system. 

Thank you very much for your help!

nakarlsson, 2021-08-07:
I don’t know SSC that well, so I can’t help you there.

You can download the specifications at ETG download section. Both CiA 402 and the modulare profiles.

It all depends on the devices you’ll implement. Reading the specs might give you some understanding on what you realy want.

leadermik, 2021-08-08:
I have found the problem. As it seems the EEPROM config must be written on the slaves every time there is an update on the object dictory. I though that would automatically happen during the flashing process (download firmware on the slave device). 😑  Just a very stupid beginner error. 

Anyway 😒 . Thank you so much for your suggestion! 😇 

So the way to go is really to read the specs and implement everything as is it descripted? Do you have a sample application, where I get some inspiration? 😮 

Thanks!

nakarlsson, 2021-08-10:
You often doesn't need to update the EEPROM, there are a few key elements there but they is often more static.
SynManager settings, Product Code, Vendor ID.

No, I don't have a good sample. And to be honest, I'm not sure I fully understand what the problem is and what you want to achieve, hence I suggest you read the specs and other input.

leadermik, 2021-08-10:
Thanks for your answer.

The board  (F2838 TI Board) I am using actually comes bundled with SSC Tool and their own slave framework, yet is very similar to yours. They are making an initial check of how big are the input & output PDOs length based on the flashed code & the config flashed inside the EEPROM. So if I change something on the PDO inputs / outputs I have to flash both. PDOs are fixed and not configurable on the fly (until based on my understanding).

The problem I am facing must actually very simple (or at least arleady solved in the industry)

We have 12 slave. How do we assign the PDO IDs? For example:

On the Inverter example we have:
1. 0x6000 Inverter_Status Input PDO
2. 0x6001 Inverter_Control_Feedback Input PDO

How do I assign them over the 12 Inverter Slave board? 

Inverter Nummer 1:
- 0x6000
- 0x6001

Inverter Nummer 2:
- 0x6010
- 0x6011 

The solution until now was to create a object dictionary excel for each one of them => parse it => flash the board => update the EEPROM. Moreover if the position on the network change then I have to repeat this process.

Is this the right way, or can I flash all of the slaves with the same PDO assigments? I am asking because, when I connected 4 bought inverters they automatically assigned PDO with this scheme. First 00, second 10, third 20 ...

The docs of Beckhof are really not great. There are hundreds of pdfs explaining a piece of the puzzle everytime and searching the docs is a nightmare. I have came accross to the CiA & modular device dictionaries, but also most 0 examples and the parsing with thier Tool (SSC) simply does not work correctly. 

Here you can take a look of the object dictionary that I using.  [inverter.xlsx](https://github.com/OpenEtherCATsociety/SOES/files/6961735/inverter.xlsx). Some examples use the 0x6nnx but when I parse it I always get 0x600x regardless of what I am defining in 0xF000.02 . What am I making wrong? 

It is really frustrating and the time for my thesis is running out.

🙏🏿 Give your advice. Thank you very much!

nakarlsson, 2021-08-12:
Hi,

What you see is what we/ETG got, the topic is quite advanced and we(rt-labs) would provide a customer with some kind of "professional services project" to help them solve this topic. 

So, I think you need to figure it out yourself or contract someone that can provide such project. 
Or, wait if some other user on GitHub have a ready to use project that fit your description.

regards
Andreas

leadermik, 2021-08-17:
Thanks the information. I will then close this issue.

closed 2021-08-17 by leadermik