OpenEtherCATsociety/SOES #104: Conformance test tool problem

AlexCorrado, 2021-09-21:
Goodmorning to all,
I'm using the conformance test tool ( CTT Beckhoff) to debug my ethercat device. This device implements SOES on a STM32 F4 and I have done a personalized driver for the LAN9252. If I run the CTT tool, it return an error in the first test block

[image]

It seems that the error is connected to register 0x134, which is called ESCREG_ALERROR and represents the AL Status code register. From the Beckoff documentation "https://download.beckhoff.com/download/document/io/ethercat-development-products/ethercat_esc_datasheet_sec2_registers_2i9.pdf", in particular page 33, it is clear that in order to cancel the "device identification" it is necessary to reset bit n°5 (counting from 0). Since this register can be rewritten only when the simulation mode is active (in my case it is not active ), it is nothing more than a copy of the AL Control register (0x120). So I tried to set bit °5 of this register to 0, but the problem was not solved. Can anyone help me?
I enclose further images regarding this error.

[image]

[image]

[image]

After this first test I reset the device and I run the other tests of CTT tool and it has returned another error, related to the state machine. In particular, the state transitions from OP mode to any other mode (e.g. PREOP, SAFEOP, etc...), I report below some images about it. I do not know if this error can be related to the previous one.

[image]

[image]

Another information that can be usefull is that with the Twincat (Beckoff) the device go to OP state and if I change manually the state the device does no go in error and the Twincat run without errors. I do not understand where the problem is, can be a SOES problem?

Thanks.

nakarlsson, 2021-09-22:
1. Can you post the ESI? SOES don't implement Explicit Device Identification, from CTT it looks like you have enabled support for it in ESI?
2. Could you post the verbose information of first testcase "#37" . Seems the state machine doesn't react on the state request to Init, is the slack stack still running? Any interrupt issue missing the AlCtrl request?

nakarlsson, 2021-12-16:
@AlexCorrado , can this be closed?

nakarlsson, 2022-06-09:
closed due to inactivity

closed 2022-06-09 by nakarlsson