OpenEtherCATsociety/SOES #111: SOEM slave esc_init error in native linux machines?

SakshamAnzo, 2021-12-27:
HI Everyone

I have a query regarding the process of setting up the slave configuration in my system.

is it necessary to have EtherCAT slave controller to setup the EtherCAT interface or to configure the device as slave using SOES?

I am currently running ubuntu linux on two generic laptops.I am trying to setup the ethercat interface inbetween these two laptops with single ethernet connectivity(one as a master and one as a slave).Will it work?...or does our typical machines have a ESC already inbuild?

Thank You

nakarlsson, 2021-12-27:
You’ll need an ESC ( EtherCAT slave Controller) on the slave side.

SakshamAnzo, 2021-12-28:
Alright! Thank for your response

nakarlsson, 2021-12-28:
If you don’t got any follow-up questions , please close the issue

closed 2021-12-28 by SakshamAnzo