OpenEtherCATsociety/SOES #114: soes for ROS2

KennyRAS, 2022-03-28:
Dear Reader,

So i want to simulate an ethercat connection between device 1 and a motor_driver by using two computers. Computer 1 will have a SOEM_forROS2 library that supposedly can make a master_node. 
But is it possible that Computer 2 can connect to Computer 1 with EtherCAT?
If so, do i need a specific library or can i just use SOEM library?
and can I connect to Computer 1, even if Computer 1 is running SOEM like it is in ROS2?

I was thinking of sending a very basic message, so that i can test if it works. But i'm kinda stuck where to go from here.
Could give someone a general guideline?

Thanks for reading my problem.

Kenny

nakarlsson, 2022-03-30:
EtherCAT slaves are run on dedicated HW, the ESC (EtherCAT Slave Controller). For SOEM running on Computer 1 to be able to talk to something on Computer 2 ujsing EtherCAT would require some ESC device on in Computer 2.

There might be ways forward, like having some SPI device on a Linux machine beeing Computer 2. 
Or
Using the FC1100 from Beckhoff on a  Windows PC.

On those kind of devices you can run for example SOES acting as the "Motor" slave.

nakarlsson, 2022-03-30:
To connect SOEM on Computer1 to Computer 2 not using EtherCAT would mean implementing a new transport layer.

nakarlsson, 2022-06-09:
@KennyRAS , can we close this?

KennyRAS, 2022-06-09:
Yes we can close it.
Thnx for all the help!

I've learned a lot about ROS and communication protocols.

closed 2022-06-09 by nakarlsson