OpenEtherCATsociety/SOES #118: Read out PDOs that are not directly adressed to the slave

MarcelloDavis, 2022-06-07:
Hello,

I am asking myself if it is possible for the slave device to read out PDOs that are not directly adressed to it by the master device?
Since the EtherCAT frame is transferred through all devices and sent back to the master, I thought that it may be possible to access the data on the fly?
Or does one have to configure the slave device on the EtherCAT-master in order to be able to access any PDO?

Thanks in advance!

nakarlsson, 2022-06-09:
The slave doesn't have access to the frame, it can only access the DPRAM. For logic LRW PDOs that access is programmed via the FMMU by the master and the FMMU is read-only from the PDI, so you can't set it up from the slave.

nakarlsson, 2022-06-15:
@MarcelloDavis , can we close this issue?

MarcelloDavis, 2022-06-18:
> @MarcelloDavis , can we close this issue?

Yes, and thank you very much for your help!

nakarlsson, 2022-08-19:
we'll close this, reopen or write a new if further questions

closed 2022-08-19 by nakarlsson