OpenEtherCATsociety/SOES #121: How do I build the SOES executables?

markmenge, 2022-06-23:
Could someone please add to SOES readme how to get started build the executables? I bought a RPI4, and was able to easily get the master built, and a program called demo for the slave (using the SOEM instructions). When I run both, nothing happens.
I spent hours and hours trying to compile other demos, but as a MS Visual Studio user I failed.

Maybe copy and paste and add the build instructions from here:
 https://github.com/OpenEtherCATsociety/SOEM#readme

nakarlsson, 2022-06-27:
Do you have an EtherCAT slave hardware connected to the RPi?

markmenge, 2022-06-27:
Not right now, although I could. I was under the impression the SOES repo was for slave demos, is that not correct?

markmenge, 2022-06-29:
First, thank you for your time to help an absolute beginner. I'm new to building C/C++ programs with the raspberry RPi. I want to learn how build the SOES slave demo programs for the RPi. I found no getting started instructions. Is there only one slave demo program?

nakarlsson, 2022-07-06:
Hi,

We'll need to update the readme, it lacks information as you highlight. For the SOES RPi DEMO **applications/raspberry_lan9252demo** you can do

1. Follow the instructions for adding support for "Broadcom BCM 2835" use by the RPi
http://www.airspayce.com/mikem/bcm2835/index.html
2. sudo apt install build-essential cmake git
3. git clone https://github.com/OpenEtherCATsociety/SOES.git
4. cmake -B build -DRPI_VARIANT=ON
5. cmake --build build

jinjiahuangsha, 2022-07-07:
hello nakarlsson !

as the newcomer , I have the similar question on how to run a demo.   

I  have IGH ec master installed.   what I suppose is,  by running an application of SOES and connecting to ec master, then I can scan and see an ec slave in master side.    By connecting an real ethercat slave card, i can see slave config shown successfully,   while by running "demo" under  "linux_lan9252demo" and connecting to master,  i see nothing in master side.     is there any step I miss or such expectation  is total incorrect ?

thanks ~~

nakarlsson, 2022-08-19:
@jinjiahuangsha , you need the real HW. 

If you don't use real HW, then you would need to create your own payload layer for transfering data from master/slave via , ex. memory. This would need modifcation in both master / slave side when it comes how the send/ receive data.

nakarlsson, 2022-08-31:
Can we close this issue?

markmenge, 2022-08-31:
I am hoping that the build instructions which exist for the master: https://github.com/OpenEtherCATsociety/SOEM/blob/master/README.md

would also exist for the slave programs. Is there a reason similar instructions are not here also?
https://github.com/OpenEtherCATsociety/SOES/blob/master/README.md

nakarlsson, 2022-09-01:
It is on the TODO list.

Mean while https://github.com/OpenEtherCATsociety/SOES/issues/121#issuecomment-1175789968 instructions should work for you

nakarlsson, 2022-10-03:
@markmenge . can we close this issue?

nakarlsson, 2022-10-03:
Have you tried https://github.com/OpenEtherCATsociety/SOES/issues/121#issuecomment-1175789968 as I asked?

markmenge, 2022-10-03:
Yes please close

nakarlsson, 2022-10-03:
We’re preparing a new release, I’ll make sure  the read me is updated

closed 2022-10-03 by nakarlsson

Amir7RN, 2023-06-07:
Hi, I followed steps in https://github.com/OpenEtherCATsociety/SOES/issues/121#issuecomment-1175789968, however, as I ran cmake --build build, I got plenty of errors such as:

home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c: In function 'bcm2835_spi_write_32':
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:67:14: error: conversion from 'int' to 'char' may change value [-Werror=conversion]
67 | data[1] = ((address >> 8) & 0xFF);
| ^
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:68:14: error: conversion from 'int' to 'char' may change value [-Werror=conversion]
68 | data[2] = (address & 0xFF);
| ^
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:72:14: error: conversion from 'uint32_t' {aka 'unsigned int'} to 'char' may change value [-Werror=conversion]
72 | data[6] = ((val >> 24) & 0xFF);
| ^
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c: In function 'bcm2835_spi_read_32':
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:84:14: error: conversion from 'int' to 'char' may change value [-Werror=conversion]
84 | data[1] = ((address >> 8) & 0xFF);
| ^
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:85:14: error: conversion from 'int' to 'char' may change value [-Werror=conversion]
85 | data[2] = (address & 0xFF);
| ^
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:92:27: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
90 | return ((data[6] << 24) |
| ~~~~~~~~~~~~~~~~~~
91 | (data[5] << 16) |
| ~~~~~~~~~~~~~~~~~
92 | (data[4] << 8) |
| ~~~~~~~~~~~~~~~^
93 | data[3]);
| ~~~~~~~~
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c: In function 'ESC_read_pram':
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:145:71: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
145 | _spi_write_32(ESC_PRAM_RD_ADDR_LEN_REG, (ESC_PRAM_SIZE(len) | ESC_PRAM_ADDR(address)));
| ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~

/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:161:22: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
161 | remainder = len - quotient4;
| ^~~
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:165:21: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
165 | quotient = (len + first_byte_position)/4;
| ^
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:166:22: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
166 | remainder = (len + first_byte_position) - quotient4;
| ^
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:172:20: error: conversion from 'int' to 'uint8_t' {aka 'unsigned char'} may change value [-Werror=conversion]
172 | fifo_range = MIN(quotient,16);
| ^~~
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:185:43: error: conversion to 'size_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
185 | buffer = (uint8_t *)realloc(buffer, size);
| ^~~~
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:191:23: error: conversion to 'size_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
191 | memset(buffer,0,size);
| ^~~~
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:197:45: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
197 | bcm2835_spi_transfern((char *)buffer, size);
| ^~~~
/home/IMU/SOES/soes/hal/raspberrypi-lan9252/esc_hw.c:202:18: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
202 | value = buffer[i] | (buffer[i+1] << 8) | (buffer[i+2] << 16) | (buffer[i+3] << 24);
| ^~~~~~
AND SO ON