OpenEtherCATsociety/SOES #124: FOE_fwrite is unable to detect a failure if write_function fails to write on last data.

fsugai, 2022-08-18:
I think there is a problem that FOE_fwrite() cannot detect failure if write_function() failed writing on last data.

The following code checks if FOE_fwrite() can write the entire data correctly.
https://github.com/OpenEtherCATsociety/SOES/blob/fc72fbeeecc10d53c9c0655c88d43214c4ef0759/soes/esc_foe.c#L473-L477

However, at line 165, ncopied is incremented even if foe_file->write_function() returns a failure flag.
This only matters if the last data write fails.
https://github.com/OpenEtherCATsociety/SOES/blob/fc72fbeeecc10d53c9c0655c88d43214c4ef0759/soes/esc_foe.c#L154-L166

I suggest the following fixes.
```
       if(failed)
       {
          DPRINT("Failed FOE_fwite ncopied=%d\n", ncopied);
       }
       else
       {
          ncopied++;
       }
```

nakarlsson, 2022-08-19:
It make sense, can you provide a pull request?

fsugai, 2022-08-19:
Thanks for the checking. I sent a PR #126.

closed 2022-08-20 by fsugai