OpenEtherCATsociety/SOES #125: How to know index PDO.

tuhai711, 2022-08-18:
Hi,
I have a problem that in the process of processing PDO, How to know the index of PDO. 
I for example: I use a converter from canopen to ethercat, then the packet includes index + lengt+data, how do I convert to send to Ethercat slave.

nakarlsson, 2022-08-19:
Could you give an example for better understanding.

CanOpen and EtherCAT differs when it comes to send/receive PDOs. EtherCAT transfer all PDOs in a "single frame" without index, length and so on. SOES takes care of the pack/unpack of the PDO mapped SDOs into/from the PDO frame.  So, from and SOES application point of view, you can simply use the object dictionary defined application variables. 

https://github.com/OpenEtherCATsociety/SOES/blob/fc72fbeeecc10d53c9c0655c88d43214c4ef0759/applications/rtl_xmc4_dynpdo/main.c#L30

there you can se the usage of the PDO data, SOES unpack the Obj.LEDgroup0.LED0 from the PDO frame for you.

tuhai711, 2022-08-31:
I see. Thank you.

tuhai711, 2022-09-01:
Yes, Thank you.

closed 2022-09-01 by tuhai711