OpenEtherCATsociety/SOES #127: Dynamic PDO and objectlist

arbite, 2022-08-25:
Hi,

I've been experimenting with the dynamic PDO functionality of SOES.
In the slave_objectlist.c we define the available PDO entries, eg:
```
const _objd SDO1C12[] = 
{
  {0x00, DTYPE_UNSIGNED8, 8, ATYPE_RWpre, acName1C12_00, 8, &Obj.SM1C12.maxsub},
  {0x01, DTYPE_UNSIGNED16, 16, ATYPE_RWpre, acName1C12_01, 0x1600, &Obj.SM1C12.value[0]},
  {0x02, DTYPE_UNSIGNED16, 16, ATYPE_RWpre, acName1C12_02, 0x1601, &Obj.SM1C12.value[1]},
  {0x03, DTYPE_UNSIGNED16, 16, ATYPE_RWpre, acName1C12_03, 0x0000, &Obj.SM1C12.value[2]},
  {0x04, DTYPE_UNSIGNED16, 16, ATYPE_RWpre, acName1C12_04, 0x0000, &Obj.SM1C12.value[3]},
  {0x05, DTYPE_UNSIGNED16, 16, ATYPE_RWpre, acName1C12_05, 0x0000, &Obj.SM1C12.value[4]},
  {0x06, DTYPE_UNSIGNED16, 16, ATYPE_RWpre, acName1C12_06, 0x0000, &Obj.SM1C12.value[5]},
  {0x07, DTYPE_UNSIGNED16, 16, ATYPE_RWpre, acName1C12_07, 0x0000, &Obj.SM1C12.value[6]},
  {0x08, DTYPE_UNSIGNED16, 16, ATYPE_RWpre, acName1C12_08, 0x0000, &Obj.SM1C12.value[7]},
};
```

If I wanted to have a huge dynamic PDO, say 256 entries, is it necessary to build a huge `const _objd` for each entry?
Is it possible to generate this information programmatically, eg through pre_object_download_hook or pre_object_upload_hook?

A second question: Is it possible to to have dynamic PDOs of differing data types, that is assigned by the master?
An example might be: One startup configuration, SOEM requests `DTYPE_UNSIGNED32, DTYPE_UNSIGNED8, DTYPE_UNSIGNED16` then on the next startup SOEM might request `DTYPE_UNSIGNED8, DTYPE_UNSIGNED8`.
I guess I'm asking if the dynamic PDOs must be pre-defined in the objectlist as the same datatypes.
I hope this makes sense!

Thank you.

nakarlsson, 2022-08-31:
>If I wanted to have a huge dynamic PDO, say 256 entries, is it necessary to build a huge const _objd for each entry?

I guess it is possible.

>A second question: Is it possible to have dynamic PDOs of differing data types, that is assigned by the master?

Yes, 

ex &Obj.SM1C12.value[0] could be assigned to PDOs of different type on start-up.

OR

ex PDO config

const _objd SDO1A01[] =
{
  {0x00, DTYPE_UNSIGNED8, 8, ATYPE_RWpre, acName1A01_00, 8, &Obj.PDO1A01.maxsub},
  {0x01, DTYPE_UNSIGNED32, 32, ATYPE_RWpre, acName1A01_01, 0x00000000, &Obj.PDO1A01.value[0]},

were &Obj.PDO1A01.value[0] can be programmed with PDO index, subindex, length.

arbite, 2022-09-01:
> I guess it is possible.

Hmm, maybe my question was not very good!

My use case is - I would like to use a large dynamic PDO as an array, but I am reasonably memory constrained.
Creating a huge `const _objd` that will only point to elements in an array is not a very good use of memory for me.
To put it more simply, I do not want to create a huge `const _objd` for the large PDO.
I was originally asking if I could generate those sections that point to the large PDO array at run time, by hooking via eg pre_object_upload_hook or pre_object_download_hook?

Or, could I just omit that section of the `objd`?

Thanks again

nakarlsson, 2022-09-15:
So you want a single PDO of type ARRAY?

arbite, 2022-09-26:
Yes I think that would be right.
Would this be using something like DTYPE_ARRAY_OF_INT for example, and setting the bitlength to anything (0 most likely).

Thank you

nakarlsson, 2023-04-12:
@arbite , can we close this issue?

arbite, 2023-04-12:
Hi, yes that is fine. Thanks