OpenEtherCATsociety/SOES #128: SOEM and SOES with EVB-LAN9252-SPI and Raspberry Pi

sc610889477, 2022-08-30:
Goodmorning to all,
I'm using a MacBook as Ethercat Master and an Ethercat Slave which uses a EVB-LAN9252-SPI and a Raspberry Pi 3B. 
The slave info with -map reports the following:
```
$./slaveinfo en24 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on en24 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:LAN9252-C2000 Sample Application
 Output size: 8bits
 Input size: 8bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000004d8 ID: 0000000d Rev: 00000001
 SM0 A:1000 L:   1 F:00090044 Type:3
 SM1 A:1200 L:   1 F:00010000 Type:4
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000001 Ll:   1 Lsb:0 Leb:7 Ps:1200 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to SII :
  SM0 RXPDO 0x1A00 Byte 0
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x3101:0x01 0x08 OCTET_STR(8) Output
  SM1 TXPDO 0x1600 Byte 0
     addr b   index: sub bitl data_type    name
  [0x0001.0] 0x3001:0x01 0x08 OCTET_STR(8) Input
End slaveinfo, close socket
End program
```
I want to use SOES's [raspberry_lan9252demo](https://github.com/OpenEtherCATsociety/SOES/tree/master/applications/raspberry_lan9252demo) to control LEDs and buttons by Raspberry Pi's GPIO. The [ESCREG_ALEVENT_SM2](https://github.com/OpenEtherCATsociety/SOES/blob/fc72fbeeecc10d53c9c0655c88d43214c4ef0759/soes/ecat_slv.c#L221) parameter is used when judging whether to use the button's function([cb_set_outputs()](https://github.com/OpenEtherCATsociety/SOES/blob/8de160a85159b1a4667004a4ec927389c1bee582/applications/raspberry_lan9252demo/main.c#L33)).

**I can execute sample_test in the slave info above, but it cannot match ESCREG_ALEVENT_SM2 and cannot control buttons.**

**Below is my train of thought, please correct me if I'm wrong.**

Above my slave info SM2 doesn't seem to start, so I should edit the ESI file to configure SM2 and SM3.
I used siitool to successfully convert the xml file into a bin file, and then used SOEM's [eepromtool.c](https://github.com/OpenEtherCATsociety/SOEM/blob/master/test/linux/eepromtool/eepromtool.c) to successfully write the bin file into the slave.
Unfortunately, the SDK of EVB-LAN9252-SPI is [LAN9252_C2000_SDK_V1.0](http://ww1.microchip.com/downloads/en/DeviceDoc/50002604A.pdf), and I did not find this SDK. The XML files I found on the web did not match my EVB-LAN9252-SPI. After forcibly writing, it will become the following result.
using [LAN9252_PIC32_SDK_v1.1](https://www.microchip.com/en-us/software-library/lan9252-ethercat-sdk)/ESI Files/SPI-withGPIO-2PortMode.xml
```
$ .slaveinfo en24 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on en24 succeeded.
1 slaves found and configured.
Calculated workcounter 0

Slave:1
 Name:Microchip PIC32 Slaves
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: e00004d8 ID: 00009252 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   4 F:00010064 Type:3
 SM3 A:1400 L:   4 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 01 FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
End slaveinfo, close socket
End program
```
running slample_test
```
$ ./simple_test en24
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on en24 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 0 0 0 0
Request operational state for all slaves
Calculated workcounter 0
Operational state reached for all slaves.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
...
```
I already asked once at SOEM.
https://github.com/OpenEtherCATsociety/SOEM/issues/638

nakarlsson, 2022-08-30:
Could you share a Wireshark of the startup?

If the slave works properly, the EEPROM is only providing basic configuration of SyncManagers , product ID, vendor ID....
The ObjectDictionary over CoE would provide the PDO configuration, wireshark would indicate if that is up and running OK.

sc610889477, 2022-08-31:
Because I don't know what demo of SOEM is executed on the master to obtain the startup information of the slave.

So before getting your reply, I shared the pcap files of slaveinfo and slamptest in advance.

Wireshark capture attached as a .zip.
[slaveinfo_and_slamptest.zip](https://github.com/OpenEtherCATsociety/SOES/files/9458638/slaveinfo_and_slamptest.zip)

nakarlsson, 2022-09-01:
SOES doesn't seem to run properly. 

Follow this instructions and build the SOES RPi sample.
https://github.com/OpenEtherCATsociety/SOES/issues/121#issuecomment-1175789968

The RAW EEPROM BIN is included , file below, and can be writting using eepromtool. 
https://github.com/OpenEtherCATsociety/SOES/blob/master/applications/raspberry_lan9252demo/slave.bin

sc610889477, 2022-09-02:
I tried the SOES RPi sample installation method of [#121(comment)](https://github.com/OpenEtherCATsociety/SOES/issues/121#issuecomment-1175789968). 
After the execution, the following results appeared.
```
~/new_SOES/SOES/build/applications/raspberry_lan9252demo $ sudo ./soes-demo
Hello Main
Hello Main
^C
```

Because it passed a series of confirmations such as ESC_BYTE_TEST_OK, I think SOES demo runs properly.

I also tried to change the hook parameter of the config to add a callback in the main.c of the demo, so as to read the register of the slave(e.g. [sc610889477/SOEM@9e6eb1d](https://github.com/sc610889477/SOES/commit/9e6eb1d)). 

Then I remake it, and the result after execution is as follows:
```
~/new_SOES/SOES/build/applications/raspberry_lan9252demo $ sudo ./soes-demo
Hello Main
Hello Main
Test CallBack.
ESCvar.App.state: 1
^C
```
It seems that the SPI communication is normal, and the LED connected to the raspberry GPIO also lights up.
___
I tried the above slave.bin and the output write as below should be successful.
```
$ ./eepromtool en24 1 -w slave.bin
SOEM (Simple Open EtherCAT Master)
EEPROM tool
ec_init on en24 succeeded.
1 slaves found.
Slave 1
 Vendor ID        : 00001337
 Product Code     : 000004D2
 Revision Number  : 00000000
 Serial Number    : 00000000
Busy.
Total EEPROM write time :757ms
End, close socket
End program
```
Then I restarted the slave, and running slaveinfo on the master produced the following results.
```
$ ./slaveinfo en24 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on en24 succeeded.
1 slaves found and configured.
Calculated workcounter 0
Not all slaves reached safe operational state.
Slave 1 State= 1 StatusCode=   0 : No error

Slave:1
 Name:evb9252_dig
 Output size: 0bits
 Input size: 0bits
 State: 1
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00001337 ID: 000004d2 Rev: 00000000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00000024 Type:3
 SM3 A:1180 L:   0 F:00000020 Type:4
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 23 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
End slaveinfo, close socket
End program
```
This doesn't seem to be successful, because the length of SM2 and SM3 is only 0. And there is no output of PDO mapping.

Finally, I tried the sample_test and found that slave couldn't enter the operational state.
```
$ ./simple_test en24
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on en24 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 0 0 0 0
Request operational state for all slaves
Calculated workcounter 0
Not all slaves reached operational state.
Slave 1 State=0x01 StatusCode=0x0000 : No error

Request init state for all slaves
End simple test, close socket
End program
```

The following is the pcap, slaveinfo, and sampletest when the bin file is written.

nakarlsson, 2022-09-05:
To me it looks like the SPI is not working properly. SOEM write to 0x120, Al control, and it succeed only the first time. After that  the WorkingCounter is always 0 pointing to SOES isn't running properly, since SOES should read 0x130, Al status, that would Ack the 0x120 write, making it possible to write 0x120 again from SOEM.

Could you add an esc_read of a know ESC register to make sure, say

12.14.6 RAM SIZE REGISTER
Offset: 0006h Size: 8 bits
Bits Description ECAT
Type
PDI
Type Default
7:0 Process Data RAM Size
This field details the process data RAM size included in the EtherCAT slave controller. The device provides 4KB.
RO RO 04h

sc610889477, 2022-09-06:
Thank you so much for your reply that made me realize what my problem was!

I found out that I made a big mistake. When running SOEM's sample_test, I didn't even start the SOES demo.
(After I confirmed slave_objectlist.c some time ago, I realized that the SOES demo will assign the frame to the corresponding PDO object area and map it to the corresponding local applications. I didn't even start the demo, and I'm still investigating why the slave_info does not display the PDO mapping intelligence.)

After running the SOES demo, I got the complete information after running slave_info.c!

```
$ ./slaveinfo/slaveinfo en24 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on en24 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:evb9252_dig
 Output size: 8bits
 Input size: 8bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00001337 ID: 000004d2 Rev: 00000000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   1 F:00010024 Type:3
 SM3 A:1180 L:   1 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000001 Ll:   1 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 23 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x01 BOOLEAN      LED0
  [0x0000.1] 0x7000:0x02 0x01 BOOLEAN      LED1
  [0x0000.2] 0x7000:0x03 0x01 BOOLEAN      LED2
  [0x0000.3] 0x7000:0x04 0x01 BOOLEAN      LED3
  [0x0000.4] 0x7000:0x05 0x01 BOOLEAN      LED4
  [0x0000.5] 0x7000:0x06 0x01 BOOLEAN      LED5
  [0x0000.6] 0x0000:0x00 0x02
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0001.0] 0x6000:0x01 0x01 BOOLEAN      Button0
  [0x0001.1] 0x6000:0x02 0x01 BOOLEAN      Button1
  [0x0001.2] 0x6000:0x03 0x01 BOOLEAN      Button2
  [0x0001.3] 0x6000:0x04 0x01 BOOLEAN      Button3
  [0x0001.4] 0x6000:0x05 0x01 BOOLEAN      Button4
  [0x0001.5] 0x6000:0x06 0x01 BOOLEAN      Button5
  [0x0001.6] 0x0000:0x00 0x02
End slaveinfo, close socket
End program
```
After changing the value of IOmap[0] in sample_test.c, I can control the LED to turn on or off!

---
However, I'm having some more issues with the buttons.
When I press the button connected to the Rasberry Pi GPIO26, the value of IOmap[1] does not change, it is always 0.

On the sample_test.c side, I use printf("%02x\n", *ec_slave[0].inputs); after [wkc = ec_receive_processdata(EC_TIMEOUTRET); ](https://github.com/OpenEtherCATsociety/SOEM/blob/f9db4f2b23455e83d9871568e3a69ab124e0c1a0/test/linux/simple_test/simple_test.c#L90)to confirm the result. Its values ​​all show 0.

On the demo side of SOES, I used printf("Button0: %d \n", Obj.Buttons.Button0); after [Obj.Buttons.Button0 = bcm2835_gpio_lev(GPIO26);](https://github.com/OpenEtherCATsociety/SOES/blob/8de160a85159b1a4667004a4ec927389c1bee582/applications/raspberry_lan9252demo/main.c#L36) to confirm the result. After running sample_test.c Button0 shows 1, after pressing the button it shows 0, and then toggles between 0 and 1 with each press.

Is this a problem writing to PDO or is my printf method wrong?

The following is the pcap of sampletest.
[sampletest3.zip](https://github.com/OpenEtherCATsociety/SOES/files/9494907/sampletest3.zip)
(After every 1000 Process Data Communications, the value of IOmap[0] will change.)

nakarlsson, 2022-09-07:
Seems to be a bug in the sample, Button0 is getting overwritten. Change to Button0, Button1...

   Obj.Buttons.Button**0** = bcm2835_gpio_lev(GPIO26);
   Obj.Buttons.Button**0** = bcm2835_gpio_lev(GPIO19);
   Obj.Buttons.Button**0** = bcm2835_gpio_lev(GPIO13);
   Obj.Buttons.Button**0** = bcm2835_gpio_lev(GPIO06);
   Obj.Buttons.Button**0** = bcm2835_gpio_lev(GPIO05);
   Obj.Buttons.Button**0** = bcm2835_gpio_lev(GPIO22);

sc610889477, 2022-09-07:
Wow! It worked! Thank you very much!
I edited Obj.Buttons.Button0 many times and didn't notice it at all.
I have resolved my issue, close this issue.

closed 2022-09-07 by sc610889477