OpenEtherCATsociety/SOES #129: FC1100 support?

jarmeister99, 2022-08-30:
Hello, I am trying to implement a Linux-based EtherCAT slave using the Beckhoff FC1100 PCIE EtherCAT Controller. This library supports the ET1100 which is on the FC1100, and I was wondering if this library provides easy support for the FC1100 as well?

nakarlsson, 2022-08-30:
I guess so, but don't ypu need a PCIE driver? On windows it is provided long the FC1100, don't know if they got a Linux drievr?

jarmeister99, 2022-08-30:
I'm not sure if there is a FC1100 Linux driver. Would I have to create my own FC1100 Linux driver? And how would SOES be integrated in this process? Thank you for the reply.

nakarlsson, 2022-08-31:
If Beckhoff doesn't provide one you would need to write your own. I don't got the details on this FC1100 device, but given your description I assume it would act as any PCIE endpoint and to support an ET1100 it would map the ET1100 address space to the PCIE address space for the Linux host/RootComplex to read. The PCIE MSI(-X) interrupts would should support the PDI IRQ and SYNC0/SYNC1 IRQ.

Given the above, SOES would be integrated in the same way as with any other ESC, SOES simply would need an esc_hw layer to do esc_read/esc_write of the PCIE address space that would represent the ESC address space. I assume one would like the PCIE driver to expose a user space API that SOES could be implemented towards?

jarmeister99, 2022-09-02:
Thank you so much, this is the route I took. I appreciate all the advice.

closed 2022-09-02 by jarmeister99