OpenEtherCATsociety/SOES #130: Connect with SOES using SOEM

avaiable-away, 2022-09-14:
➜  linux_lan9252demo git:(master) sudo ./demo
Hello Main
Hello Main

➜  simple_test sudo ./simple_test ens33 
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on ens33 succeeded.
No slaves found!
End simple test, close socket
End program

The slave SOES cannot be found by the SOEM.

What configuration should I do to make SOEM connect to SOES？
Thanks！

nakarlsson, 2022-09-14:
SOES rely on havning dedicated EtehrCAT HW

How does your setup look like?

avaiable-away, 2022-09-14:
Is the hardware necessary? I think SOES can be used as an ECAT slave simulator.
Maybe I was wrong.

nakarlsson, 2022-09-14:
>Is the hardware necessary? I think SOES can be used as an ECAT slave simulator.
>Maybe I was wrong.

Not out of the box.

avaiable-away, 2022-09-14:
> > Is the hardware necessary? I think SOES can be used as an ECAT slave simulator.
> > Maybe I was wrong.
> 
> Not out of the box.

Thanks！

When I use the hardware connection, the SOEM can connect to the slave station.
➜  simple_test sudo ./simple_test ens33
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on ens33 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 8 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
Processdata cycle 2480, WKC 3 , O: 00 00 I: ff 00 00 40 71 04 T:716^Z2914636652521

I want to know where SOES uses testing.

nakarlsson, 2022-09-15:
Any more questions or can we close this issue?

avaiable-away, 2022-09-15:
> Any more questions or can we close this issue?

Is there a tutorial for using SOES?
You can close it now. Thank you.

nakarlsson, 2022-09-15:
There is a chapter on SOES in the rt-labs EtherCAT-SDK manual

https://rt-labs.com/refman/ethercat-sdk_refman.pdf

closed 2022-09-15 by nakarlsson