OpenEtherCATsociety/SOES #131: Issue creating STRING using ethercat-sdk

bfrost831, 2022-09-15:
I'm not sure if this is the correct place for issues regarding the ethercat-sdk tool, but I am trying to create string variables in both RXPDO and TXPDO for use with a custom EtherCAT slave built on SOES. I am creating string variables using the ethercat-sdk selecting the "VISIBLE_STRING" datatype, and this correctly creates a char array in `utypes.h`. The other STRING data types behave the same, but according to [this documentation](https://download.beckhoff.com/download/document/io/ethercat-development-products/an_et9300_v1i8.pdf) "VISIBLE_STRING" seems to be the correct option.

However, when I upload the binary to the EEPROM of my slave device and scan from a Beckhoff PLC, these variables are empty. I cannot link a STRING variable from my MAIN program of the same size to these variables. When I double click on this variable in Twincat, it shows no name, type, size, etc.

Should I do something else to create strings in PDO?

nakarlsson, 2022-09-15:
Send the issue to support[AT]rt-labs.com

nakarlsson, 2023-04-12:
Close since it should be handled by rt-labs

closed 2023-04-12 by nakarlsson