OpenEtherCATsociety/SOES #132: How to send PDO from SOES to SOSM?

tuhai711, 2022-09-15:
Hello,
I have a bropblem, about PDO, SOSM only send (ec_send_processdata) then received (ec_receive_processdata) message from slave. How to slave auto send PDO (input/oputput) to master?

nakarlsson, 2022-09-15:
That is how EtherCAT works, all traffic is driven by the master.

tuhai711, 2022-09-15:
Yes. Thank you very much,

tuhai711, 2022-09-15:
That mean, Slave can not notify PDO to Master, that is right?

nakarlsson, 2022-09-15:
Not without the Master sending a frame.

tuhai711, 2022-09-15:
Yes, I see.  If Master not sending a frame, Slave can not notify PDO to Master. Thank you.

nakarlsson, 2022-09-15:
Please close "Issues" when no more questions

tuhai711, 2022-09-16:
Hello Again, Can Slave use  SDO notify to master?

nakarlsson, 2022-09-16:
No, the same goes for SDO. The master initiate read/write of the SDO.

tuhai711, 2022-09-16:
Do you have method for salve notify to master?

tuhai711, 2022-09-16:
Hello agaim, Do you know any method for salve notify to master without a master's request?

nakarlsson, 2022-09-16:
No, the slaves never send frames

nakarlsson, 2023-04-12:
Closed due to inactivity

closed 2023-04-12 by nakarlsson