OpenEtherCATsociety/SOES #141: Problem with SOES on STM32 with LAN9252

YBKIMadn, 2023-02-09:
Hi,
I am trying to run EtherCAT communication between STM32F4 using LAN9252 (via SPI). now I have very similar problem with the following git

https://github.com/OpenEtherCATsociety/SOES/issues/55

I also have Invalid mailbox header errors but the EtherCAT communication via SOEM is possible. 
I read the above git but I couldn't get the solution to fix.

Can you suggest a solution to solve this problem ?

tuhai711, 2023-02-10:
Hello, Did you changed in the file "rt-kernel-lan9252/esc_hw.c" for spi_select, write,read?

YBKIMadn, 2023-02-10:
Thank you for response! 
I didn't changed the code inside of esc.c file. 

I edit the EtherCAT code based on diebieslave git as following git web address
https://github.com/DieBieEngineering/DieBieSlave

So, I have "esc.c" file not the "rt-kernel-lan9252/esc_hw.c" and I didn't edit this file at all
Can you give advise to edit this file to spi_select, write, read?

nakarlsson, 2023-04-12:
@YBKIMadn , is this still an issue or can we close it?

nakarlsson, 2023-05-04:
close due to inactivity

closed 2023-05-04 by nakarlsson