OpenEtherCATsociety/SOES #145: How to use SOES without emulated EEPROM

elsp1991, 2023-02-22:
I recently started working in a new ethercat slave project and I am struggling to understand how to use the stack without emulated eeprom. Basically my ESC has its own eeprom and should be able to deal with it, so the library doesnt need anymore to implement the EEP_Write EEP_read function in HAL but I am not sure how to remove these functions.

My guess is that in the ecat_config struct I have to leave the .esc_hw_eep_handler function pointer to NULL and the ESC itself would take care to process the request. In that case is easy but how can I acess the EEPROM from the application now?

ArthurKetels, 2023-02-24:
Your guess is right. Just NULL the .esc_hw_eep_handler. 

Depending on your ESC hardware there needs to be an EEP_read() and EEP_write() function implemented. Look in your ESC datasheet how to do that. 

But many slave applications do not need to bother about the EEPROM because the interface is mostly used by the master side.

elsp1991, 2023-02-27:
I can confirm my that setting NULL with a compatible ethercat slave controller the stack works fine the ESC is taking care to serve the EEPROM requests from the master

elsp1991, 2023-02-27:
I am still digging in the ESC protocol to see if the application can have an abstracted(emulated or not) access to eeprom through the standard ESC protocol

andarm, 2023-04-06:
Hi @ArthurKetels, 
If want to read the default value from eeprom, how the know the relation of xml and eeprom?  
Seems it is hard code in source code. That is confuse me.  Why not read eeprom match with the code?
i.g. 
[image]
[image]

nakarlsson, 2023-04-07:
Why do you need to read the EEPROM? In the picture above you refer to the ObjectDictionary, you can read that via the Obj struct and the .value member.

andarm, 2023-04-08:
Thanks for your responnse. 
If xml has changed, my code have to fixed again. If can xml mapping in EEPROM and read it as configure file that would convinence.  Yes, via Obj struct it is working. That seems xml work for Master only, as the slave xml just for reference.  That is my confuse.

nakarlsson, 2023-04-08:
That is how EtherCAT works. The XML for master to support Offline configuration.  The OD and EEPROM for online configuration

The OD for the slave application.

In general the slave doesn’t make use of the EEPROM. Some data is used during startup to write specific ESC registers, that is handled by the ESC.

So, if the application change, you should do a firmware upgrade to reflect that change.

nakarlsson, 2023-04-08:
For parameters in the OD you can mark them as Backup And/Or Setting to support handling of persistent data. The store to flash or similar is implemented by the slave application.

If you spend some time reading the EtherCAT standards I think you’ll find your answers.

andarm, 2023-04-08:
Got it, thanks @nakarlsson .

nakarlsson, 2023-04-09:
Om further questions open a new issue

closed 2023-04-09 by nakarlsson