OpenEtherCATsociety/SOES #151: SOES, SOEM and dcsync

arbite, 2023-04-19:
Hiya,

I'm after a clarification of the order of initialisation for dcsync with SOES and SOEM.
The comment on this issue https://github.com/OpenEtherCATsociety/SOEM/issues/520#issuecomment-852809564 says that we should be calling ec_configdc() in PRE-OP, then calling ec_dcsync0() later in SAFE-OP, just before transitioning to OP. 
I also see that the SOEM ebox.c example does not call ec_dcsync0() until we are actually in OP state and that the master has sync'd to the DC.

If I have a dig around in SOES esc.c, I see that ESC_checkDC() is called from ESC_startinput(), when transitioning to SAFE-OP. ESC_checkDC() looks at the SYNC_ACT register, which is programmed by ec_dcsync0() (SOEM), which, if following the SOEM examples and guidance will not be set at the transition to SAFE-OP (only after in the transition to OP, etc). 

This means that SOES dcsync will not be configured properly if we follow the examples for SOEM.
Can someone please confirm that this is the case? And that it is acceptable to call ec_dcsync0 (SOEM) all the way back before we transition to SAFE-OP?
Or, even better - Hopefully I'm actually doing something wrong and someone can correct me...

Thanks

ArthurKetels, 2023-04-23:
Probably I should retire the ebox.c example as it is very old and written for a very specific piece of hardware. At the same time a new DC example should be written. Still, for now you should take https://github.com/OpenEtherCATsociety/SOEM/issues/520#issuecomment-852809564 and https://github.com/OpenEtherCATsociety/SOEM/issues/487#issuecomment-786245585 as reference. 

In general as a slave you should be as lenient as possible with regard to the master synchronization and start-up sequence. In the ebox hardware you could switch from SM to SYNC0/1 in any state, change the cycle time during OP, etc.etc. It simply tracked the current situation and did not enforce anything. The paradigm being, whatever the master instructs me is law. 

The current batch of slaves out in the world today behave like pedantic teenagers that all have their own view of what is acceptable and not. Try to be the master of that!

So to answer you question. A SOEM master application should enable SYNC0/1 just before the transition from pre-OP to safe-OP. A SOES slave application should check SYNC0/1 after it received the safe-OP to OP transition request and before setting OP state. 

The only exception being when it is necessary for total application that the master can check proper SYNC0/1 acceptance of all slaves before safe-OP to OP transition. Then the slave should check SYNC0/1 after the pre-OP to safe-OP transition. This is for example the case when the slave provides inputs to the master control system that are synchronization critical. Input data from the slave is active at safe-OP after all.

solomon112, 2023-04-25:
hi
   I use linux-lan9252 demo.  TwinCAT3 is ERR PREOP。can not in Safe-Op

nakarlsson, 2023-04-25:
@solomon112 , write your own Issue instead of hi-jack another

arbite, 2023-04-25:
Hey Arthur, thanks for your reply.
I have things working well with enabling dcsync just before safe-OP.
For my interest, if I was to aim to make my slave more lenient, so that it was fine with enabling dcsync, for example, during OP,
what would be the correct SOES way of achieving this?

nakarlsson, 2023-08-09:
ESC_checkDC allow you to override the basic check "ret = ALERR_DCINVALIDSYNCCFG;" by providing a esc_check_dc_handler that simpy return "OK".

I'll close the issue, reopen or create a new issue if needed. Thanks for understanding.

closed 2023-08-09 by nakarlsson