OpenEtherCATsociety/SOES #162: Does the EtherCAT slave ESC process messages in store-and-forward mode?

Lambo0604, 2023-10-21:


nakarlsson, 2023-10-25:
No, it process data on-the-fly.

Lambo0604, 2023-10-27:
Can you explain the data processing of SDO? My understanding is that SDO requires the assistance of the MCU. In theory, messages should be stuck in the MCU before being forwarded

nakarlsson, 2023-10-27:
That is handled by the SOES stack, running in the MCU. You need to study how EtherCAT works. 

Here is a quick start. https://www.ethercat.org/download/documents/EtherCAT_Device_Protocol_Poster.pdf

Lambo0604, 2023-10-30:
For PDO data, if running in SM synchronization mode, once PDO messages are received, MCU interrupt is triggered to read or write PDO datas; If there is a processing delay of several tens of us on the MCU side, forwarding of this PDO message will delay several tens of us； I think the processing of SDO will also limit the processing speed of MCU; So  I'm confused about how to on-the-fly

nakarlsson, 2023-10-30:
No, the RxPDO/TxPDO will be processed by the ESC right away via SyncManagers.
The MCU can act on the RxPDO in the interrupt and read the newly updated RxPDO data in the SM.
The on-the-fly mean the TxPDO will be added to the frame when passning through. While the TxPDO produced by the MCU interrupt will be sent one cycle later. 

Ex:
Frame 1
Frame RxPDO(1) -> SM2
Frame TxPDO(0) <- SM3

IRQ1 
SM2 -> RxPDO(1)
SM3 <- TxPDO(1)

Frame 2
Frame RxPDO(2) -> SM2
Frame TxPDO(1) <- SM3

IRQ2
SM2 -> RxPDO(2)
SM3 <- TxPDO(2)

nakarlsson, 2023-11-02:
1. No, if you check the poster above it is different.
2. Om errors the frame get flagged with an error nibble in the end and data ignored by the ESC. Following ESCs Will detect the Error nibble and ignore the data. The frame will not get dropped by the ESC. All described the standard.

closed 2023-11-02 by nakarlsson

Lambo0604, 2023-11-03:
Thanks a lot~