OpenEtherCATsociety/SOES #163: Wildcards in foe filenames

n-sterenborg, 2023-11-02:
First off, hats of to you guys for this repo, fantastic work.

In the current implementation, the filename of a foe file that is send to the slave has to match (one of) the predefined filename(s) exactly. It would be nice to be able to use wildcard characters for this. My application is for a firmware update, it would be nice to be able to specify e.g.: "firmware_v*.*.*.*.bin".

ArthurKetels, 2023-11-02:
Easy, just specify the part before '*' as the filename in SOES, extra characters will be ignored. It is up to you application to check the file for validity.

n-sterenborg, 2023-11-03:
Hi @ArthurKetels, thanks for your reply. I tried your solution but it does not seem to work for me.

I specified the filename in the foe_cfg as "firmware_v". If I then send a file named e.g. "firmware_v1.2.3" foe_fopen() returns FOE_ERR_NOTFOUND, since the strncmp compares up to num_chars, which is the length of "firmware_v1.2.3".

ArthurKetels, 2023-11-04:
Changing `#define FOE_FN_MAX      31` to a value that fits "firmware_v" helps. But I agree the API could be improved so the application has better control about what filenames can be accpted or not.