OpenEtherCATsociety/SOES #182: Entirely simulated slave device under Linux

lanouettea, 2024-04-30:
In our hardware solution, we use a motor drive that is controlled via Ethercat. A PC acts as the master using the SOEM library and communicates with the drive. 

To test our software stack, we have simulators for most of our hardware. This allows us to perform end to end testing of our products from our CI system and to develop more efficiently by simulating the hardware on developer PCs. However, we are struggling to find a way to simulate an Ethercat device. We already have all the logic to control a simulated motor drive and all the logic to send data to the Ethercat device from our software using SOEM, but we cannot figure out how to implement a simulated Ethercat slave. 

We are wondering if SOES could be used to achieve this. The simulated hardware will never end up in the final product. We believe we should be able to create 2 virtual network interfaces (one for the master, one for the slave) and bridge them together. We cannot figure out how to actually use that virtual network interface to send and receive data from SOES or if it is even possible. 

Any help or pointer to the right direction would be greatly appreciated. 

Thanks!

nakarlsson, 2024-05-01:
It is possible to get something runniing.
In another project we did something similar, the trick is to decide the frame on the SOES side, but that is not that difficult, in the end the frames only different ways to access the memorial of the slave. 

The input to SOES is ESC memorial registers, and you would need to emulate the PDI IRQs that triggers state changes, update SMs and so on.

maximekli, 2024-06-25:
Hi @lanouettea, very interested on that subject as well, any success on your side?

nakarlsson, 2025-01-02:
Close due to inactivity

closed 2025-01-02 by nakarlsson