OpenEtherCATsociety/SOES #183: Slave writing to 7000 PDO outputs

lagrecjo, 2024-06-19:
My 7000 output PDOs are updating correctly from the master. However, some of the 7000 PDOs can be changed on the embedded slave via other means (Keypad, UART console interface, etc). If the value mapped to the 7000 PDOs is changed via means other than the master it is overwritten by RXPDO_update(), the local change is lost, and the target always reflects the state of the PDOs from the master, even though no change was initiated by the master.
Is there any way to have the master observe (read) the state of the 7000 output PDOs and not change them unless specifically written by the master? It seems the 7000 PDOs are 'write-only' opposed to 'read-write'.

nakarlsson, 2024-06-19:
Sounds like you should have a local copy updated by the 7000 Object. Then you can decide when to update.

It is rarly that the 7000 object is the end that is the output.

Ex. You have a GPIO outputs, then you need to write that from the update cb_outputs.

GPIO_set(0x7000)

R/W for 0x8000 is access rights from the master.

lagrecjo, 2024-06-19:
Yes, thank you, that is a method for sure and what I am doing as a work-around. 

Is it true there is no way to have the master read back the state of the 7K objects? They are 'write-only'? Even if they are updated before the frame goes out? I am thinking here that the master never inspects the incoming 7K objects.

The issue with this (if true) is that if one has a running system there is no way to acquire the state of the 7K objects upon ECat connection. I was duplicating them in the 6K in order to have the state but that is quite a bit of overhead. If this is the only way then I suppose that is what I need to do unfortunately. A negative to this is that the in vs out states will not match in the master until they are updated by the master. Shocking there are no R/W objects in ECat,

_"R/W for 0x8000 is access rights from the master."_ Could you explain further?

nakarlsson, 2024-06-20:
I think your mixing concepts here. PDOs and SDOs work differently.
The SDOs are accessed using CoE mailbox transfers, having R/W properties. Read & wirtes are done explicitly.

While, PDOs are SDOs mapped to be updated by incoming frames using the 3 buffer process data SyncManagers. They than become subordinated by the processdata exhange. So if you have a SDO 7k mapped as PDOs the get updated by the master as you describe it.

Of course the master can access the 7k object as a SDO, it wan be r/w but the write using SDO will be overwritten by the current PDO value.

This is just how EtherCAT works

closed 2025-01-02 by nakarlsson