OpenEtherCATsociety/SOES #184: DIG_process(...) not called in "esc_hw.c" under "soes/hal/tiesc".

matthijs040, 2024-08-07:
Hi,

A colleague of mine is having an issue with the ESC HAL helper functions in esc_hw.c for the TI platform.
Specifically, in the `void PDI_Isr(void)`-function, the `DIG_process(...)`-invocation never happens as the &-flag-mask:
`if(ESCvar.ALevent & (ESCREG_ALEVENT_SM2 & ESCREG_ALEVENT_SM3))` never matches. 
In our fork we resolved this by changing the mask to: 
`(ESCREG_ALEVENT_SM2 | ESCREG_ALEVENT_SM3)`. 
Is this change correct? Or is there some reason for the mask being this way that we are missing?

Kind regards,
Matthijs

nakarlsson, 2024-08-07:
Yes, that make sense.

https://github.com/OpenEtherCATsociety/SOES/blob/8f3841e3cefd83c48d866ffaeb2d92c146bf3cd7/soes/hal/rt-kernel-xmc4/esc_hw.c#L246

matthijs040, 2024-08-07:
Oh, missed that. Guess I should have checked the other examples. Thanks for the confirmation. 👍

closed 2024-08-07 by matthijs040