OpenEtherCATsociety/SOES #189: SOEM/SOES XMC4300: No PDO communication

alexswerner, 2025-03-01:
Hi,

thanks everybody involved that this stack is available! For now I'm just having a little trouble to run the demo provided in applications/xmc4300_slavedemo in this repo together with simple_test from the SOEM repo. 

## Reproducing
* Flash XMC4300 with unmodified code
* Execute unmodified simple_test
```
$ sudo ./simple_test  enxf0b4d22bef07
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enxf0b4d22bef07 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 6 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
Processdata cycle    1, WKC 3 , O: 00 I: a5 09 ff 1f 53 T:794142244870072680
Processdata cycle    2, WKC 3 , O: 00 I: a5 09 ff 1f 53 T:794142244880300040
Processdata cycle    3, WKC 3 , O: 00 I: a5 09 ff 1f 53 T:794142244890572280
Processdata cycle    4, WKC 3 , O: 00 I: a5 09 ff 1f 53 T:794142244900809880
Processdata cycle    5, WKC 3 , O: 00 I: a5 09 ff 1f 53 T:794142244911067600
Processdata cycle    6, WKC 3 , O: 00 I: a5 09 ff 1f 53 T:794142244921334000
Processdata cycle    7, WKC 3 , O: 00 I: a5 09 ff 1f 53 T:794142244931593120
...
```
* Press and hold button in XMC4300 board. Observe no change of input bytes. The expectation is that input bytes change.

## Testing I/O
If I modify the code on the XMC4300 to turn on the LED when the button is pressed from cb_set_output, things work. That confirms that the this function is called and that the I/Os are correct. If I modify the SOEM/simple_test example to write to 0x7000/0x01 via SDO, this turns on the LED. From this I assume that the I/O hardware is correctly configured and many parts of the setup work correctly.

## Slave mapping
```
sudo ./slaveinfo enxf0b4d22be
f07 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enxf0b4d22bef07 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:ATO
 Output size: 8bits
 Input size: 40bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00001337 ID: 12783456 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   1 F:00010024 Type:3
 SM3 A:1180 L:   5 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000001 Ll:   5 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 13 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x08 UNSIGNED8    LED
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0001.0] 0x6000:0x01 0x08 UNSIGNED8    Button
  [0x0002.0] 0x6000:0x02 0x20 UNSIGNED32   Encoder
End slaveinfo, close socket
End program
```
This looks correct to me.

## Wireshark
If I modify the SOEM/simple_test example to send alternating values of 0 and 1 via the mapped PDO, I can observe these values in wireshark leaving the master. However, there is no change in the Input PDOs of this datagram when it's coming back from the slave. It always contains the value: `a5 09 ff 1f 53`. The working counter does increase.

## Guessing
My first guess is that this is about SYNC0 and SYNC1. I've tried to configure sync0 by adding `ec_dcsync0(1, TRUE, 10000000, 0);` right after `ec_configdc()` but this had no effect. I'm not positive that this is actually supported by the slave.

## Details:
* SOEM platform: x86_64 with USB-Ethernet adapter with Debian 12
* SOEM version: eaf523e
* SOES platform: Infineon XMC4300 Relax EtherCAT Kit **v1.1** / MCU: XMC4300F100 / PHY: BCM5241 / ModusToolBox 3.4.0 / XMClib v2.1.6p1
* SOES version: 2752dc2
* Laptop directly connected to single slave

Thanks for having a look at this issue!

alexswerner, 2025-03-03:
Hi @nakarlsson,

thanks for the quick response. I've posted the patched simple_test code to https://github.com/alexswerner/SOEM/tree/debug_pdo . For testing the SDO access I'm using this code:
https://github.com/alexswerner/SOEM/blob/eff500ef57e11f2c64087cb40d57d61e2918a1d6/test/linux/simple_test/simple_test.c#L77-L100
and there is the wireshark pcapng capture of a simple_test run [simple_test.pcapng.gz](https://github.com/user-attachments/files/19055889/simple_test.pcapng.gz). Note, this is running the code with a modification which should make the LED blink: https://github.com/alexswerner/SOEM/blob/eff500ef57e11f2c64087cb40d57d61e2918a1d6/test/linux/simple_test/simple_test.c#L123

> Does reading 0x6000 reflect the Button?

I assume reading via SDO, yes it does. See section TEST_BUTTON in the referenced code.

> The length should match the TxPDO size of 5Bytes.

When I stop at ecat_slv.c:158 in the debugger and print ESCvar.ESC_SM3_sml the value is 5 - correct.
On the master side I see that OBytes has a length of 5 as well.

> The txpdo should be Rb.button as assigned in main.c

Yes it is:
```
uint8_t * txpdo = (uint8_t *)&Rb.button;
```
and in the debugger, executing `print txpdo[0]` I can see the button status correctly.

## Futher steps
I can confirm that RXPDO_update is called on the microcontroller and that it reads from address 0x1100. I'm new to the details of ET1100 but in `EtherCAT_ET1100_Datasheet_all_v1i8.pdf` page 63 or section 8.1 Buffered Mode it states: `Buffer 1
(invisible shall not be used)`. I just trust that this code worked before.

When I break in RXPDO_update and step into ESC_read and look at the argument `buf` (memcpy dst) it's at a different address than rxpdo:
```
(gdb) p buf
$7 = (void *) 0x1fff0974 <rxpdo>
(gdb) p rxpdo
$8 = (uint8_t *) 0x1fff0900 ""
```
and indeed at this address (0x1fff0974) I can see the expected rxpdo content - the byte toggles from 0x00 to 0x01 every call. Why gdb hints that 0x1fff0974 is associated with rxpdo is beyond my knowledge.

So if I replace
```
uint8_t * rxpdo = (uint8_t *)&Wb.LED;
```
with
```
uint8_t rxpdo[1];
```
in `main.c` on the XMC4300 side, the LED can be controlled by the master. Hmm. So this is something about variable alignment. This is compile with these flags:
```
arm-none-eabi-gcc -c -DXMC4300_F100x256 -MMD -MP -MF XMC_Peripheral_Library/Newlib/syscalls.d -MT XMC_Peripheral_Library/Newlib/syscalls.o -I././XMC_Peripheral_Library/XMCLib/inc -I././XMC_Peripheral_Library/CMSIS/Include -I././XMC_Peripheral_Library/CMSIS/Infineon/XMC4300_series/Include -I. -I../../soes -I../../soes/hal/xmc4 -I../../soes/include/sys/gcc -g -Wall -fmessage-length=0 -Wa,-adhlns=XMC_Peripheral_Library/Newlib/syscalls.lst -mfloat-abi=softfp -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mthumb -Os -ffunction-sections -fdata-sections  -std=gnu99 -pipe
```
and 
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arm GNU Toolchain 11.3.Rel1) 11.3.1 20220712
```
from ModusToolbox 3.4.0.

nakarlsson, 2025-03-04:
Verify that the rxpdo/txpdo get linked correct for this setup.

https://github.com/OpenEtherCATsociety/SOES/blob/eaf523e7faecaf2c5c2143001a6f59a06c480590/soes/ecat_slv.c#L25

nakarlsson, 2025-04-12:
Close due to inactivity

closed 2025-04-12 by nakarlsson